Zolko-123/FreeCAD_Assembly4 #413: Improve fields allowing operations and also the use of formulas

leoheck, 2023-03-12:
I was looking for a way to improve some fields of the Assm4.

For instance, the animation, and the placement have simple fields. Freeecad however, has some nice fields that support operations and also this Fx field that allows using formulas keeping the formula explicit. 

I got some help on how to use these kinds of fields from Freecad, but I could not make it work yet with Asm4.

I am sharing this here because maybe you have a better understanding of how this can be added to the Asm4. 

The discussion started here. 
https://forum.freecad.org/viewtopic.php?t=76577#p665983

Zolko-123, 2023-03-12:
Yes, this is a recurring request. ATM you can enter plain numbers in the Asm4 dialogs, and when you need expressions you can do it explicitly from the property window.

Please keep in mind that one of Asm4's objectives (well, one of my objectives) is to keep code as simple as possible, and not adding as much bells and whistles as possible

leoheck, 2023-03-12:
Sure, I don't want to add any code at all.

Zolko-123, 2023-05-29:
Any news on that ???

closed 2023-05-30 by leoheck