Zolko-123/FreeCAD_Assembly4 #415: Can't edit placement - "Please select a link in the assembly model"

judge2005, 2023-03-19:
At some point this stopped working - I always get "Please select a link in the assembly model". The result is that I can't attach the model's coordinate system to the assembly's coordinate system.

I'm using FreeCAD 0.20.2 on a Mac, Assembly4 0.12.6 and Python 3.10.8

The screenshot below shows that I have an imported part selected:

[image]

Zolko-123, 2023-03-19:
1) there is no need to put all those Bodies inside Parts, you can use Bodies directly

2) this message appears if the link is not in as Asm4 Model, but since I cannot see what's up in the tree I can't say for sure

judge2005, 2023-03-20:
Yes - I didn't have a model. I guess if anything, perhaps the _insert part_ option shouldn't be enabled if there isn't a model, but otherwise I obviously forgot I had to do that from the last time I used it. Many thanks.

closed 2023-03-20 by Zolko-123