Zolko-123/FreeCAD_Assembly4 #420: Unable to Insert Part. New Model

AccidentallyTheCable, 2023-03-31:
Ive imported a board using KicadStepUp mod, and am trying to insert it as a Part in my main file. When doing so, I get the following error, and am unable to insert it. Ive tried many other models as well without success.

I even made a basic sketch, extruded it, and attempted to import it, with the same error.

`03:52:27  Initializing Assembly4 workbench (v0.12.5) ............................ done.
03:53:14  Traceback (most recent call last):
03:53:14    File "E:\Users\AccidentallyTheCable\AppData\Roaming\FreeCAD\Mod\Assembly4\.\insertLinkCmd.py", line 276, in onCreateLink
03:53:14      createdLink = self.rootAssembly.newObject( 'App::Link', linkName )
03:53:14  RuntimeError: {'sclassname': 'class Base::RuntimeError', 'sErrMsg': 'Cannot add static property', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Cannot add static property', 'btranslatable': False, 'breported': False}`

Zolko-123, 2023-03-31:
> insert it as a Part in my main file

what do you mean by that ? What versions of FreeCAD and Asm4 are you using ?

```
RuntimeError: {'sclassname': 'class Base::RuntimeError', 'sErrMsg': 'Cannot add static property', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Cannot add static property', 'btranslatable': False, 'breported': False}
```

This looks like a FreeCAD bug

AccidentallyTheCable, 2023-03-31:
Sorry, I guess I should've included that info, heh.

Tried both FreeCAD 0.20.1 and 0.20.2

Tried Asm4 versions from master all the way back to 0.9.x before I gave up.

Zolko-123, 2023-04-01:
> Tried Asm4 versions from master all the way back to 0.9.x before I gave up

then it's not an Asm4 issue, please use the FreeCAD help forum

closed 2023-04-01 by Zolko-123