Zolko-123/FreeCAD_Assembly4 #428: Import datum object not working in group

joukeHijlkema, 2023-04-21:
When I try to import a datum object from a part in an assembly that is in a group I get the following error :

```
14:43:58  <Exception> ObjectIdentifier.cpp(1533): Property 'Placement' not found in 'Group.Placement'
in property binding 'Optique3#Interface_simple_bottom001_LCS_boulon004.Placement'
14:43:58  <Exception> ObjectIdentifier.cpp(1533): Property 'Placement' not found in 'Group.Placement'
in property binding 'Optique3#Interface_simple_bottom001_LCS_boulon004.Placement'
14:43:58  Recompute failed! Please check report view.
```
This is my setup

```
OS: Ubuntu 22.10 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32929 (Git) AppImage
Build type: Release
Branch: master
Hash: 1dcc58c4b1c51c50215639ca58326f6b69867743
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.18.8
  * fasteners
  * workfeature
  * lattice2 1.0.0
  * CfdOF 1.21.7
  * ThreadProfile 1.76.0
  * Geomatics
  * workfeature-macro
  * Assembly4 0.50.1
  * sheetmetal 0.2.63
  * parts_library.bak
  * Render 2023.2.3
```

joukeHijlkema, 2023-04-21:
[image]

joukeHijlkema, 2023-04-21:
Part001_LCS_1 is imported from Part001 and works fine
Part002_LCS_1is imported from Group/Part002 and does not work

Zolko-123, 2023-04-21:
Should be fixed in the development branch

closed 2023-05-29 by Zolko-123