Zolko-123/FreeCAD_Assembly4 #435: Interferences Check discussion

leoheck, 2023-05-08:
I am experimenting with the current Interferences check in the development branch.

I have two observations.

1 ====================

I am seeing duplicated items.
[image]

Is this by choice? 
In my code, I have chosen to show just one of those, since they are equal.
If this is a choice, and since they are the same object, it would be nice to paint them with the same color, what do you think?

2 ====================

The shape of the LCS inside the Assembly is being created inside the ShapeCopies, and this is weird since it shows this plane that does not belong to the model. I believe this is not an intended behavior, can you confirm?
[image]

Zolko-123, 2023-05-08:
> 1) I am seeing duplicated items. Is this by choice?

No. I've seen that too, but I don't know where it comes from. I hoped you would find the mistake

> 2) I believe this is not an intended behavior, can you confirm?

this comes probably from `if obj1.isDerivedFrom('Part::Feature') `. This was intended to also take fasteners into account. May-be a bad choice after-all. Dunno, what do you think, should fasteners also be checked for interference ? For the time being, I'll revert to check _App::Link_ only

Zolko-123, 2023-05-08:
Also, there is something I don't understand: you make all the intersections (_make_intersection_), and then you remove those that are empty (_remove_empty_common_). Why not ignore from the beginning empty intersections ?

leoheck, 2023-05-08:
> No. I've seen that too, but I don't know where it comes from. I hoped you would find the mistake

I see, I will try to find the issue then. 

> May-be a bad choice after-all. Dunno, what do you think, should fasteners also be checked for interference ? For the time being, I'll revert to check App::Link only

I would like to have the fasteners tested too. The user could ignore them when inspecting visually since everything is transparent. I would just not include them if this becomes processing intensive. 

I have a pretty good machine and maybe small assemblies. So far I did not have any issues regarding the processing taking too much time. I also was thinking to generate each intersection with threads, but I am not sure how to make this yet. Also, being able to abort the operation would be interesting. Do you know anything about using threads with Freecad?

> Also, there is something I don't understand: you make all the intersections (make_intersection), and then you remove those that are empty (remove_empty_common). Why not ignore from the beginning empty intersections ?

I added this "remove when empty" some days later to avoid having too many objects created. My code definitely needs refinement. Maybe this is such a thing that can be tweaked.

leoheck, 2023-05-19:
I am closing this topic since I've pushed some commits to improve the Interference Check.

closed 2023-05-19 by leoheck