Zolko-123/FreeCAD_Assembly4 #439: isAssembly is checking if the model name is Assembly

leoheck, 2023-05-26:
Hi Zolko, I just saw this method above from Asm4_Libs.

```
def isAssembly(obj):
    if not obj:
        return False
    if obj.TypeId=='App::Part' and obj.Name=='Assembly':
        if hasattr(obj,'Type') and obj.Type=='Assembly':
            return True
    return False 
```

I was using this function to check if something is an Asm4 assembly. But it limits to something that is named Assembly. 

Is it important to make this test?

I would prefer to name the car I am assembling "Wheel" instead of "Assembly" which makes more sense when assembling a Car than I can add multiple "Wheels".

Zolko-123, 2023-05-29:
Yes, it is important to name it "_Assembly_" to guarantee that there is only 1 assembly per file. 

What I do is to set the _Label_ to the name I want, and this is visible in the tree. Wouldn't that work for you ?

leoheck, 2023-05-29:
Ah, true, makes sense. I made a mistake confusing Name Label. I was having some issue and I tough it was because of that. I will use it to check if it works now with the makeBom working better.

closed 2023-05-29 by leoheck