Zolko-123/FreeCAD_Assembly4 #444: 'Asm4_makeBOM' failed on fresh install

JonasThomas, 2023-07-04:
[code]
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git) AppImage
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * Assembly4 0.50.2
  * sheetmetal 0.2.63
  * fasteners 0.4.56
[/code]

```
15:10:50  Initializing Assembly4 workbench (0.50.2) ............................ done.
15:11:06  Running the Python command 'Asm4_makeBOM' failed:
Traceback (most recent call last):
  File "/home/jonasthomas/.local/share/FreeCAD/Mod/Assembly4/./makeBomCmd.py", line 91, in Activated
    file = open(ConfUserFilejson, 'r')

[Errno 2] No such file or directory: '/home/jonasthomas/.local/share/FreeCAD/Templates/Asm4_infoPartConf.json'

```
This topic is discussed further on the freecad forum:
#https://forum.freecad.org/viewtopic.php?t=79452

snelweg, 2023-10-05:
same here:

`Expecting value: line 1 column 1 (char 0)Running the Python command 'Asm4_makeBOM' failed:
Traceback (most recent call last):
  File "/home/ton/.local/share/FreeCAD/Mod/Assembly4/./makeBomCmd.py", line 92, in Activated
    self.infoKeysUser = json.load(file).copy()
  File "/tmp/.mount_FreeCA7plMcE/usr/lib/python3.10/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/tmp/.mount_FreeCA7plMcE/usr/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/tmp/.mount_FreeCA7plMcE/usr/lib/python3.10/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/tmp/.mount_FreeCA7plMcE/usr/lib/python3.10/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None

Expecting value: line 1 column 1 (char 0)`

> [code]
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34506 (Git) AppImage
Build type: Release
Branch: master
Hash: 3ce68745dc3aa3b02de0df2d5e4409bf3e286c49
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Freecad-Built-in-themes-beta 1.2.2
  * ExtremeProLight 1.6.0
  * Render 2023.8.18
  * Assembly4 0.50.2
  * FeedsAndSpeeds 0.5.0
  * POV-Ray-Rendering
[/code]

Zolko-123, 2023-12-04:
Yes, unfortunately BoM is currently broken. I don't use it and all those who jump in do it only for their own workflow

snelweg, 2023-12-05:
Can we please at least have it back working the way it was before it broke?

Zolko-123, 2023-12-05:
Yes, if you remember which version was working

snelweg, 2023-12-05:
ok, thank you!

 I will try to find a working version.

Zolko-123, 2023-12-05:
That would be very helpful

Zolko-123, 2023-12-06:
> I will try to find a working version.

I suggest you try v0.12.7 (or 6), in the changelog it seemed to work

snelweg, 2023-12-06:
> > I will try to find a working version.
> 
> I suggest you try v0.12.7 (or 6), in the changelog it seemed to work

Thanks! Will try this afternoon.

Zolko-123, 2023-12-06:
You can also try the development branch, I just merged BoM code that was lurking there for 6 months, may-be it works.

snelweg, 2023-12-06:
> You can also try the development branch, I just merged BoM code that was lurking there for 6 months, may-be it works.

Thank you!
So the development appimage will soon contain the 'old' BOM code?

Zolko-123, 2023-12-06:
euh ... no. You have to manually download the [development](https://github.com/Zolko-123/FreeCAD_Assembly4/tree/development) branch, unzip-it, and move it to the Mod directory. It is all described in the [instructions](https://github.com/Zolko-123/FreeCAD_Assembly4/blob/master/INSTRUCTIONS.md).

snelweg, 2023-12-06:
> euh ... no. You have to manually download the [development](https://github.com/Zolko-123/FreeCAD_Assembly4/tree/development) branch, unzip-it, and move it to the Mod directory. It is all described in the [instructions](https://github.com/Zolko-123/FreeCAD_Assembly4/blob/master/INSTRUCTIONS.md).

Aha, you meant your dev branch. Ok, clear. I will do that.
Thanks for helping me out. Keep up the good work!