Zolko-123/FreeCAD_Assembly4 #447: You cannot have a variable named like one of the supported measurement units.

madhead, 2023-08-09:
Imagine having these variables in an assembly:

[image]

If you try to use some of them, the expression parser will fail:

[image]
[image]

While the other works perfectly:

[image]

It seem like the expression parser doesn't like any variables or aliases to be named like one of the [supported units](https://wiki.freecad.org/Expressions#Units). I guess it happens somewhere near the `ExpressionParser::isTokenAUnit`.

It would be good to see a cleaner explanation in this case, other than the "Failed to parse expression".

Zolko-123, 2023-08-09:
you can't be serious in claiming this

madhead, 2023-08-09:
@Zolko-123, excuse me?

This is what really happens and I had to read the source code (the original FreeCAD) to understand why my `t` variable (which is a suitable name for a time, for example) just does not work. This is frustrating. And I imagine not every FC user would be able to understand.

This issue could not happen, for example with Spreadheets, as they just don't allow a user to use units as aliases. But the A4 spreadsheet allows users to create variables that could not be referenced later. I don't understand why it's an invalid issue.

Zolko-123, 2023-08-09:
> I don't understand why it's an invalid issue.

tough, you have a lot to learn

madhead, 2023-08-09:
Then maybe link to a doc where it's described?

Zolko-123, 2023-08-09:
A doc that says that " _kg_ " is a reserved variable name ? If you're such an ignoramus please don't even try to use Assembly4

madhead, 2023-08-09:
The only doc I found and linked above says (it's not a A4 doc, though)

>> If you have a variable whose name is that of a unit you must put the variable between << >> to prevent it from being recognized as a unit. For example if you have the dimension Sketch.Constraints.A it would be recognized as the unit ampere. Therefore you must write it in the expression as Sketch.Constraints.<<A>>. 

That trick doesn't work with A4.
[image]

And while using a `kg` may sound idiotish to you, there is no reason not to make a product more user-friendly if it is possible.

Zolko-123, 2023-08-10:
t ≠ kg

closed 2023-08-10 by Zolko-123

madhead, 2023-08-10:
Well, FC recognizes `t` as a tonne:

[image]

Here is a link to their parser or so: https://github.com/FreeCAD/FreeCAD/blob/79a31d7483e78d3facf6c58fd6479165819c6776/src/App/ExpressionParser.l#L212-L213

So, `t` is a unit too. Just like many other letters and combinations like `N` or `A` or others.

madhead, 2023-08-10:
And in my original issues there were both. Both `kg` and `t, and `test`. Just to demonstrate the issue.

madhead, 2023-08-10:
`oz`, not being a SI init, though, would not work as a variable too.

madhead, 2023-08-10:
Something like #448 might prevent this class of exceptions and make your workbench a little bit less frustrating.

Zolko-123, 2023-10-01:
Merged #448 , after all it's not a bad idea, thank-you.