Zolko-123/FreeCAD_Assembly4 #449: Circular Array SegFault

9R, 2023-08-22:
I have a FCStd that I designed some time last year using circular arrays in assembly 4 that now seams to be broken because of some recent changes in assembly4.

If I checkout commit 26a36cec1d906e9f34655f1ff154e1fa41424034 tagged v0.12.7 I can still open and edit the file.

With the next commit e4ea02aed0dff74e2cef8f4a43389fbeaadc8db5 tagged v 0.50.0 FreeCAD segfaults when I try to open the file.

Error message:
```
<class 'AttributeError'>: Module Asm4_objects has no class CircularArray
<class 'AttributeError'>: Module Asm4_objects has no class CircularArray
<class 'AttributeError'>: Module Asm4_objects has no class CircularArray
<class 'AttributeError'>: Module Asm4_objects has no class CircularArray
<class 'AttributeError'>: Module Asm4_objects has no class CircularArray
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x39ab0) [0x7fdecd24fab0]
#1  0x7fdecf8aebb4 in App::PropertyLink::getValue() const from /usr/lib/freecad/lib/libFreeCADApp.so+0x4
#2  0x7fdecf86b0bc in App::LinkBaseExtension::syncElementList() from /usr/lib/freecad/lib/libFreeCADApp.so+0x44c
#3  0x7fdecf86bc9d in App::LinkBaseExtension::update(App::DocumentObject*, App::Property const*) from /usr/lib/freecad/lib/libFreeCADApp.so+0xa8d
#4  0x7fdecf872a67 in App::LinkBaseExtension::onExtendedDocumentRestored() from /usr/lib/freecad/lib/libFreeCADApp.so+0x5a7
#5  0x7fdecf7decec in App::DocumentObject::onDocumentRestored() from /usr/lib/freecad/lib/libFreeCADApp.so+0x4c
#6  0x7fdecf7800af in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /usr/lib/freecad/lib/libFreeCADApp.so+0x76f
#7  0x7fdecf780b35 in App::Document::afterRestore(bool) from /usr/lib/freecad/lib/libFreeCADApp.so+0xc5
#8  0x7fdecf90f720 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /usr/lib/freecad/lib/libFreeCADApp.so+0x13f0
#9  0x7fdecf910801 in App::Application::openDocument(char const*, bool) from /usr/lib/freecad/lib/libFreeCADApp.so+0x121
#10  0x7fdecf9198af in App::Application::processFiles(std::__cxx11::list<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /usr/lib/freecad/lib/libFreeCADApp.so+0x13f
#11  0x7fded03d9a8c in Gui::MainWindow::delayedStartup() from /usr/lib/freecad/lib/libFreeCADGui.so+0x18c
#12  0x7fdecdac3d74 in QObject::event(QEvent*) from /usr/lib/libQt5Core.so.5+0x234
#13  0x7fdece77893f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8f
#14  0x7fded0048841 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x91
#15  0x7fdecda9c2f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x128
#16  0x7fdecdaa125b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt5Core.so.5+0x19b
#17  /usr/lib/libQt5Core.so.5(+0x2e72d8) [0x7fdecdae72d8]
#18  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x141) [0x7fdecbd0fa31]
#19  /usr/lib/libglib-2.0.so.0(+0xb7cc9) [0x7fdecbd6ccc9]
#20  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x32) [0x7fdecbd0d0e2]
#21  0x7fdecdaeb15c in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x6c
#22  0x7fdecda9b004 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x134
#23  0x7fdecda9c4a3 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x93
#24  0x7fdecffc30c4 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x2064
#25  FreeCAD(+0x3990) [0x559470aea990]
#26  /usr/lib/libc.so.6(+0x23850) [0x7fdecd239850]
#27  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x7fdecd23990a]
#28  FreeCAD(+0x4a55) [0x559470aeba55]
```

I get the same behavior with the current tips of master & development branches.

I have tested this with FreeCAD 0.20.2 and 0.21.0.

System: Arch Linux
File exhibiting the bug: [circularArraySegFault.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/12409872/circularArraySegFault.zip)

Can assembly4 be fixed or is there any way to repair my project file to work with the current version assembly4?

Zolko-123, 2023-08-24:
oooooops ... yes, the array tool has changed – for the better – recently, and I didn't check retro-compatibility. 

For the sake of improvement, can you please delete the old array and try with the new tool, and report here your findings ? Then we can see the best way forward

Zolko-123, 2023-08-24:
> File exhibiting the bug: [circularArraySegFault.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/12409872/circularArraySegFault.zip)

as a side-note: that is a strange way to use a circular array

Zolko-123, 2023-08-24:
I tried to play with your file but it's quite confusing how you built it. The best I can suggest for a fast solution is to open it with the Assembly4 version you have created it with, delete all circular arrays, then re-open with the new Assembly4 version and re-create the arrays with the new tool. 

Unfortunately, you have nested arrays quite deeply in the model so it's not trivial for an external to know how it has been built

9R, 2023-08-25:
Thanks for your quick response.
Yes, I know the design might be a bit unorthodox, but it was a nice way of using the angular symmetry of the object to skip a lot of repetitive work.
Luckily, after taking a closer look at my design I was able to solve the issue in the way you suggested [^details] , since the 4 circular arrays causing the issue where just applied to 4 separate object with no further operation depending on the resulting arrays.

Resulting project file: [circularArrayFixed.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/12443589/circularArrayFixed.zip)

Unfortunately I ran into another issue, now with variables. This I could not resolve by deleting & recreating. Background: I use multiple variables, some containing expression engine formulas to animate the (un-)folding mechanism of the design.

When I open the `Animate Assembly` dialog I now get this error message:

```
Traceback (most recent call last):
  File "/home/foo/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 221, in onSelectDoc
    self.updateVarList()
  File "/home/foo/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 206, in updateVarList
    animationHints.cleanUp(self.Variables)
  File "/home/foo/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 819, in cleanUp
    hint = hintList.get(entry, None)
           ^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
```
When I select the `time` variable in the `Animate Assembly` dialog, this error message follows and I can not set the range, step size, etc or animate the assembly.
```
Traceback (most recent call last):
  File "/home/foo/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 235, in onSelectVar
    aniHints = animationHints.get(self.Variables, selectedVar)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/foo/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 798, in get
    return hintList.setdefault(varName, defaultHints)
           ^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'setdefault'
```

I get the same behavior even after deleting and recreating all variables.

If I open the segfaulting project file from the [comment above](https://github.com/Zolko-123/FreeCAD_Assembly4/issues/449#issue-1861634825) with assembly commit tagged `v0.12.7` everything works as expected. The fixed file from this comment obviously does not animate with the `v0.12.7` but also does not animate with `master`

I could not find any obvious differences in the object containing variables in my new file when comparing it to a freshly created file initialized as assembly4 (I only looked in the GUI, have not started comparing `Document.xml`, yet).

Do you have any hints/ideas how I could fix this?

PS: Sould I open a separate issue for the variable problem, since the original issue with segfaulting arrays is kind of resolved?

[^details]: Just in case someone runs into a similar problem as me, here are the steps how to repair a project that segfaults with arrays after updating assembly4 to > v0.50:
  in `FreecadModPath/Assembly4`:
  `git checkout v0.12.7`
  create backup just in case:
  `cp /path/to/my/projectfile{,_bak}.FCStd`
  open project
  `FreeCAD /path/to/my/projectfile.FCStd`
  deleted all instances of circular arrays
  save & close
  `git checkout master`
  `FreeCAD /path/to/my/projectfile.FCStd`
  recreate all instances of circular arrays
  save & close

9R, 2023-08-30:
I did some further testing. The animation problem is unrelated to my old file but also happens in newly created files and objects when using assembly4 ≥v0.50.

Since the original issue here was resolved and by recreating the arrays I am closing this issue and will file a new bug report with proper description to how to reproduce it in a separate issue: #450.

closed 2023-08-30 by 9R