Zolko-123/FreeCAD_Assembly4 #450: Animation with Variables broken in ≥ v0.50.0

9R, 2023-08-30:
There seems to be an issue with animation since the update to ≥ v0.50.

When animation a linked object in assembly4 everything works as expected. But when the file is saved, closed and opened again assembly animation is broken. When pressing the `Animate Assembly` button, the dialog does not open and a error is displayed in the console:

```
11:41:00             ^^^^^^^^^^^^^^^^^^^
11:41:00  AttributeError: 'NoneType' object has no attribute 'setdefault'
11:41:00  Running the Python command 'Asm4_Animate' failed:
Traceback (most recent call last):
  File "/home/foo/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 154, in Activated
    self.updateVarList()
  File "/home/foo/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 209, in updateVarList
    self.onSelectVar()
  File "/home/foo/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 235, in onSelectVar
    aniHints = animationHints.get(self.Variables, selectedVar)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/foo/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 798, in get
    return hintList.setdefault(varName, defaultHints)
           ^^^^^^^^^^^^^^^^^^^

'NoneType' object has no attribute 'setdefault'
```

### Steps how to reproduce:

  * Initialize assembly 4 object on an empty file, save, close, reopen
  * Create a body object `b_test`
  * Insert object into assembly model attached to LCS_Origin
  * Create Variables
    * m1_start =10
    * m1_dur =10
    * m1_end = 20 [^note]
    * time = 0
[^note]: At first I had an expression here that calculated `m1_end` with `m1_start + m1_dur` and thought this might be the cause, but the behavior is the same regardless of having a scalar or an expression in this variable
    * time = 0
  * Set Attachment offset for an axis (i.e. z) to `Variables.time < Variables.m1_start ? 0 : 20 * (Variables.time - Variables.m1_start) / Variables.m1_dur`
  * Save file
  * Open `Animate Assembly` dialog, select Variable `time`, set reasonable range limit & move the slider
=> animation works ✅
  * Close file, open file, click on `Animate Assembly` button
=> not dialog window + error message in report view ❌

### Video demonstrating the behavior:

[Video](https://github.com/Zolko-123/FreeCAD_Assembly4/assets/381298/34dd7803-37e9-4014-888b-2d33095d4534)

### FCStd exhibiting the bug:
[assembly_animation_bug.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/12474708/assembly_animation_bug.zip)

Under some conditions the `Animate Assembly` dialog still opens but no variables can be selected. At the moment I can not reproduce this behavior on a fresh file.

### Software used:
System: Archlinux
FreeCAD: v0.21.0
Assembly: v0.50.2

9R, 2023-10-17:
I ran some tests after the latest changes. The issue still persists.

9R, 2023-12-17:
Just ran some tests with v0.50.6: The issue is gone.
Users might have to recreate animations from old, pre-v0.50 assemblies in a newly initialized assembly.
Closing the issue.

closed 2023-12-17 by 9R

ohazi, 2023-12-26:
I'm still seeing this issue in Debian testing (FreeCAD 20.2 and assembly4 0.50.6). Not sure whether anything in the slightly older version of FreeCAD is having an effect, but I can test this again on my arch laptop in a few days. 

I've applied the following snippet from this forum post: https://forum.freecad.org/viewtopic.php?p=700080#p700080

```
diff --git a/AnimationLib.py b/AnimationLib.py
index 8529e0c..e8699d0 100644
--- a/AnimationLib.py
+++ b/AnimationLib.py
@@ -804,6 +804,10 @@ class animationHints():
         if "AnimationHintList" not in variables.PropertiesList:
             variables.addProperty("App::PropertyPythonObject", "AnimationHintList", "AnimationHints", "The hintfield for the animation dialog").AnimationHintList = {}
             variables.setPropertyStatus("AnimationHintList", "Hidden")
+
+        if variables.getPropertyByName("AnimationHintList") == None:
+            variables.AnimationHintList = {}
+
         return variables.getPropertyByName("AnimationHintList")
```

which (so far) appears to make the issue go away.