Zolko-123/FreeCAD_Assembly4 #451: [Assembly4] Mirrored objects are always in black

bjnobre, 2023-09-14:
I can't change the color of a a mirrored object on assembly 4.
[See the image with the error here](https://gitlab.com/bjnobre/shared2web/-/raw/main/WhatsApp_Image_2023-09-11_at_13.36.00.jpeg?ref_type=heads)

bjnobre, 2023-09-14:
OS: KDE Flatpak runtime (qtile)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at f670854)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * sheetmetal 0.3.0

virgiliu, 2023-10-04:
Happens to me too on Windows 10. See video for repro.
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * fasteners 0.4.66
[/code]

https://github.com/Zolko-123/FreeCAD_Assembly4/assets/426097/4994dd86-90bd-4be8-b0c9-dac8faf116fd

bjnobre, 2023-10-04:
I've found a solution!
See image bellow (set SHOW ELEMENT property to true).
[solution](https://i.imgur.com/2LPOEuF.png)

closed 2023-10-04 by bjnobre