Zolko-123/FreeCAD_Assembly4 #456: Q/FR: How do I re-access the "place linked part" dialogue?

LibrarySeph, 2023-11-28:
> Today is my first day messing around with Assembly 4. Please consider this onboarding user testing, I'm clearly not an expert. Maybe someday I will be, but I'll only ever be a novice once.

I wanted to modify an existing assembly attachment, and couldn't find a way to open the original menu:

[image]

I did find the menu below, but it is not as easy to navigate:

[image]

### FR:

If there isn't a way to get back to that menu, I would like to suggest making it possible through an item similar to "constraints" in sketch mode.  I do see there is already a constraints folder, and that's where I'd expect to find a place to modify this.
[image]

Zolko-123, 2023-12-02:
> How do I re-access the "place linked part" dialogue ?

did you try to select the object you wanted to change and click on the 
[image] icon ? 

Or did you try to read the [Instructions](https://github.com/Zolko-123/FreeCAD_Assembly4/blob/master/INSTRUCTIONS.md) ?

LibrarySeph, 2023-12-04:
I did peruse the instructions.  I did not try clicking that icon.

Upon selecting an Assembly child, this is the tasks view:
[image]

After clicking the place link icon, this is the task view:
[image]

Is there any reason why the empty task view couldn't show this information by default?  It's empty at the moment.

Zolko-123, 2023-12-04:
So it all works as expected. What should the task view show if there are no tasks going on ?

closed 2023-12-04 by Zolko-123

LibrarySeph, 2023-12-04:
If one selects the part, it could show the place linked part dialogue.  Or perhaps a list of potential common options, like using the place link button.  It would help with discoverability quite a bit.

Also, I think it's okay for there to be a difference between "works as expected" in regards to someone who's learning and someone who knows the tool through-and-through.  I'm only one person, as are you.  It takes a lot of people expressing their expectations before trends can be found.  I do not expect my input alone to suit that, but sharing it might help register a valuable shift in user flows sooner.