Zolko-123/FreeCAD_Assembly4 #466: Mirrored object > techdraw plot >>> bad mirror on page

bjnobre, 2024-02-13:
When "show element" property of a mirror object is not set to true, the object is mirrored object is shown on black and the plot on a techdraw page is correct:
[image]

But when I set the property to true, so I can see mirrored object on color, it shows rotated (perhaps) on techdraw page:
[image]

But when I set the property to true, so I can see mirrored object on color, it shows rotated (perhaps) on techdraw page:
[image]
[image]

zisoft, 2024-03-03:
To see the mirrored object in its original color, open the mirror group and select the original part.
Then switch to the "View" tab, right-click --> Show All

In the "Object Style" group find "Lighting" and set it back to "One side"

bjnobre, 2024-04-03:
Hi!

There is no such group or property.
[image]

zisoft, 2024-04-03:
You have selected an array, try to select the object below.

[image]

Zolko-123, 2024-06-03:
What is the situation on this subject ?

bjnobre, 2024-06-03:
It is solved.

closed 2024-06-03 by bjnobre