Zolko-123/FreeCAD_Assembly4 #467: Insert Part Dialog in Assembly4 0.50.7,  FreeCAD 0.21.1, on Win10 Obscures Error Message and Locks UI

alecjames, 2024-02-16:
Environment:
FreeCAD Version: 0.21.1
Assembly4 Version: 0.50.7
Operating System: Windows 10
Description:
When attempting to insert a part into a new, unsaved document using Assembly4, an error message is generated to inform the user that the document needs to be saved first. However, this error message is obscured by the 'Insert Part' dialog, which seems to have a top-most property, making the error message inaccessible. The 'OK' button of the error message box is hidden behind the 'Insert Part' dialog, and the cancel button on the 'Insert Part' dialog becomes inoperative as the dialog loses focus. This results in the application appearing unresponsive.

Steps to Reproduce:

Launch FreeCAD 0.21.1 with Assembly4 0.50.7 on Windows 10.
Try to insert a part into a new, unsaved document.
Expected Result:
An accessible error message prompts the user to save the document before inserting parts.

Actual Result:
The 'Insert Part' dialog obscures the error message, causing the application's UI to lock up. (The 'Insert Part' dialog's cancel button does not function since its not in focus).

Workaround:
Pressing the escape key closes the obscured error message,

Zolko-123, 2024-02-19:
> this error message is obscured by the 'Insert Part' dialog, which seems to have a top-most property, making the error message inaccessible. The 'OK' button of the error message box is hidden behind [...]

strange, on my system the error message is on top:

[image]

alecjames, 2024-02-19:
I'm on Windows Version 10.0.19045.4046 but in any case the insert part dialog has "topmost" -

[image]

Zolko-123, 2024-06-03:
What is the situation with this ?

closed 2024-06-07 by Zolko-123