Zolko-123/FreeCAD_Assembly4 #475: asm4 breaks file index in combo tree view

wolfgangr, 2024-03-13:
Symptom:

After working some time with asm4 (still early tests....), when I try to close files using the menue or <<ctrl+W>>, different files get closed than the one selected in the Combo tree view.
I think it is related when asm4 opens additional files automagically.

System info:
```
OS: Debian GNU/Linux 10 (buster) (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.5.13
  * slic3r-tools
  * Pyramids-and-Polyhedrons
  * A2plus 0.4.64
  * Assembly3 0.12.2
  * Assembly4 0.50.9
  * FeedsAndSpeeds 0.5.0
  * dodo 1.0.1
  * frame 0.1.1
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * MOOC 2022.4.21
  * Help 1.0.3
  * ose-piping
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.2
  * ThreadProfile 1.89.0
  * SteelColumn
  * Estimate 0.1.2
  * DynamicData 2.60.0
  * parts_library
  * btl 0.9.9
  * AnimationFreeCAD 1.0.0 (Disabled)
```

wolfgangr, 2024-03-14:
encountered the situation again, right now. Will try to add **reproducibility** info:

Have two asm4 assemblies, **one including the other as variant part.**
My design is still errorneous - tried to close the included Variant, then this error occurs.
To me, it looks like the **file stays open** [which may be OK due to asm4 part dependency], but some internal **index is rebuilt as if it had been successfully closed**.

After that, the relation between the tree view and the files appears corrupt.

error console:
```
09:10:39  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: list index out of range
```

python console:
```
>>> ### Begin command Std_SaveCopy
>>> # Gui.SendMsgToActiveView("SaveCopy")
>>> App.getDocument("walker_link_new").saveCopy("/home/wrosner/cleo-wrosner/daten/hardtec/CNC/FreeCAD/tutorials/parametric/asm4/walker_link_straight.FCStd")
>>> ### End command Std_SaveCopy
>>> ### Begin command Std_SaveCopy
>>> # Gui.SendMsgToActiveView("SaveCopy")
>>> App.getDocument("walker_link_new").saveCopy("/home/wrosner/cleo-wrosner/daten/hardtec/CNC/FreeCAD/tutorials/parametric/asm4/walker_link_triang.FCStd")
>>> ### End command Std_SaveCopy
>>> ### Begin command Std_SaveCopy
>>> # Gui.SendMsgToActiveView("SaveCopy")
>>> App.getDocument("walker_link_new").saveCopy("/home/wrosner/cleo-wrosner/daten/hardtec/CNC/FreeCAD/tutorials/parametric/asm4/walker_link_disc.FCStd")
>>> ### End command Std_SaveCopy
>>> # Gui.runCommand('Std_CloseActiveWindow',0)
>>> # Gui.runCommand('Std_CloseAllWindows',0)
>>> # Gui.Selection.clearSelection()
>>> # MRU="2"
>>> # with open('/tmp/.mount_freecaOaEw7L/usr/share/Mod/Start/StartPage/LoadMRU.py') as file:
>>> # 	exec(file.read())
>>> # App.setActiveDocument("walker_leg_new")
>>> # App.ActiveDocument=App.getDocument("walker_leg_new")
>>> # Gui.ActiveDocument=Gui.getDocument("walker_leg_new")
>>> # Gui.Selection.addSelection('walker_leg_new','Spreadsheet')
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('walker_leg_new','Assembly','kinematic_master.')
>>> # Gui.Selection.clearSelection()
>>> # import Show
>>> # ActiveSketch = App.getDocument('walker_leg_new').getObject('kinematic_master')
>>> # tv = Show.TempoVis(App.ActiveDocument, tag= ActiveSketch.ViewObject.TypeId)
>>> # ActiveSketch.ViewObject.TempoVis = tv
>>> # if ActiveSketch.ViewObject.EditingWorkbench:
>>> #   tv.activateWorkbench(ActiveSketch.ViewObject.EditingWorkbench)
>>> # if ActiveSketch.ViewObject.HideDependent:
>>> #   tv.hide(tv.get_all_dependent(App.getDocument('walker_leg_new').getObject('Assembly'), 'kinematic_master.'))
>>> # if ActiveSketch.ViewObject.ShowSupport:
>>> #   tv.show([ref[0] for ref in ActiveSketch.Support if not ref[0].isDerivedFrom("PartDesign::Plane")])
>>> # if ActiveSketch.ViewObject.ShowLinks:
>>> #   tv.show([ref[0] for ref in ActiveSketch.ExternalGeometry])
>>> # tv.sketchClipPlane(ActiveSketch, ActiveSketch.ViewObject.SectionView)
>>> # tv.hide(ActiveSketch)
>>> # del(tv)
>>> # del(ActiveSketch)
>>> # 
>>> import PartDesignGui
>>> # ActiveSketch = App.getDocument('walker_leg_new').getObject('kinematic_master')
>>> # if ActiveSketch.ViewObject.RestoreCamera:
>>> #   ActiveSketch.ViewObject.TempoVis.saveCamera()
>>> #   if ActiveSketch.ViewObject.ForceOrtho:
>>> #     ActiveSketch.ViewObject.Document.ActiveView.setCameraType('Orthographic')
>>> # 
>>> # Gui.runCommand('Std_CloseActiveWindow',0)
>>> 
```

Zolko-123, 2024-03-15:
I'm not sure to understand, but here it is:

1) if you link a part into the assembly ("Insert Part") then the corresponding document (file) is opened in the background. If you save the assembly, it will also want to save the other documents of the linked parts, that's normal.

2) for variant links, this is somewhat different, because the corresponding document is not a file but a temporary hidden document (in RAM). So if you close this hidden document by hand, I don't know what happens exactly in the background. If everything fails then the best is to close and reopen FreeCAD

wolfgangr, 2024-03-15:
hm ... thanks, anyway, but sounds like "living with the quirks" ...

Is it advisable to switch to a development build to get the last finish?  Is the 0.22 appImage OK? I  tried to build FreeCAD from source, but my old debian10 is missing loads of dependecies.

I really like the concept of Asy4 - to know what I do. But still it takes some dozens more time to get simply models running than it would cost me to draft a skript in OpenSCAD...

I'd like to contribute to the maturity of asy4 by beta testing, if this helps. Maybe write some documentation as well.

wolfgangr, 2024-03-15:
> if you link a part into the assembly ("Insert Part") then the corresponding document (file) is opened in the background. 

may be that points to the cause what happens. I have both my current assembly and my corresponding parts documents open and visible in the tree. When I try to close one of the parts documents, FreeCAD refuses to do so - or maybe reopens it again - fine with that. However, some index in the GUI seems to me like "assuming" that the closing action was succesful- and show different content than the one I click on.

I just remember that most time I encountered these problems I had spreadsheets open as well. Maybe this changes the internal relations of the file, window, viewport, whatever indexing.

wolfgangr, 2024-04-10:
My guess is it is a part issue. 
They accept 2D parts as models in 0.21.2, while the wiki for 0.21.1 still states that only one 3D part can be selected.
Looks like I have to join the bleeding edge.

Sadly, the deveolopment AppImage is built without Assembly4.
Are you aware of some git fork where **Assembly 4 and the path development are kept in sync?**

Zolko-123, 2024-04-10:
> They accept 2D parts as models in 0.21.2, while the wiki for 0.21.1 still states that only one 3D part can be selected

I don't understand what you mean, sorry. Who is "they" ?

wolfgangr, 2024-04-10:
sorry for using sloppy language in what looks like mined ground ;-O
https://forum.freecad.org/viewtopic.php?style=2&t=86133

I'm new to FreeCAD and quite even so to 3D-CAD and CAM.

I assume that there are some people collaborating in developping Path / CAM, maintaining it in the development mainstream and writing the wiki for it. That's what I wanted to refer to as "they".

I don't want to join whatever quagmire.
I'd just like to use Assembly4 and the new developments in Path/CAM together.
If that were possible....

Zolko-123, 2024-04-10:
I didn't want to embarrass you or ask you to take sides, sorry for the confusion, it was a genuine question.

To be honest, I don't think that Path/CAM should work directly on an assembly, that doesn't make sense it the real world, you don't manufacture an **assembly**, you manufacture **parts**. An assembly will contain, by definition, parts, bodies, and links to parts and bodies, therefore it's the Path WB that should check its inputs, and not throw a random error because it doesn't understand what the user asks.

What I do is to export the part that I want to make into a new document, and work with the Path WB on that

[image]

wolfgangr, 2024-04-11:
Maybe I use a different style to make things, or did not find the proper way of modeling it in FreeCAD.

Just have a look at this picture (and the process following)
https://forum.freecad.org/viewtopic.php?p=748649#p748649

the bundles of al-profiles represent the X and the Z axis of a ttc450 hobby mill.
Turned out to be not so sturdy as the ads told, so I attached linear rails to it.
So I took models of the profiles and the rails, and assembled them (using Assembly4) into correct position.

Now the final job is to actually connect the carriages sliding on the X-Axis to the Z-Axis.
For that, I milled two pieces aluminum L_profiles.
Put a piece into the assembly (well, defined in a spreadsheet) and matched first the position, then the holes, to the parts that have to be connected.

I'm an agricultural engineer by profession, and that's the way I think. 
Have a tractor and a plough share - need a frame to connect them.

Have an elevator, a silo and an limit angle - need a pipe to feed the grain into the silo.
Have pipe and a piece of roof, need a piece to fix it.
But well, the height of the elevator depends on the angle of the pipe, and the roof has to cover all that...

So I think backwards from the purpose, to come down to the parts required.
In contrast - starting from the parts (Elevator -> roof -> pipe x grain flow angle -> silo) requires an iterative process.
Working backwards ends in "first time fits" (ideally....).

Till now, my constructions were not so complicated, 2D and hands on testing with welder and angle grinder worked fine.
Enough slabs of timber to hold the parts in place, till the steel grows towards them...
And of course, the silly mill rail bracket could have been designed the other way round.

Zolko-123, 2024-04-11:
> So I think backwards from the purpose, to come down to the parts required.
> In contrast - starting from the parts (Elevator -> roof -> pipe x grain flow angle -> silo) requires an iterative process.
> Working backwards ends in "first time fits" (ideally....).

This is called top-down design (function => assembly => part) and Assembly4 was made for that process. While most traditional assembly workflows need the reverse (part => assembly, function is lost)

But that doesn't exclude to use Path separately: when I want to manufacture a part made in this way, I still export the final part to a separate document and use the Path WB on that document. It's true that you lose interactivity, but that's also the case in real life: once a part is manufactured, you cannot change its length anymore. The advantage to export a part for manufacturing is that you freeze the design at that stage (we call it "as-built"), and don't run into the danger of changing the assembly after manufacture which would give you a model that doesn't correspond to reality.

wolfgangr, 2024-04-11:
Agree. In the end, this is basically a question of style, taste, and living with what reality allows us.
Just wanted to answer your question why I wanted to mill from assembly parts.

Basically, the very dimensions both for positioning the LCS and for drilling holes come from the same spreadsheet.
(have been thrown from the learning curve ladder by flipping sketches, toponame issues and bloatware stuff not working serveral times)
Just working on enhancing the Z-Axis as well. Same game, in principle.
Created the part in the main assembly, opened a new "mill-part-assembly" where I can match both symmetric copies to a single fixture, and make simple part cuts for processing. "Part cut" cuts parametric dependency I think that's what you call "as built".

wolfgangr, 2024-04-11:
> I didn't want to ... ask you to take sides

mmmhh ... not much impact required, actually ... 
Honestly, I don't like to see me on skd of "side" - prefer to stay back or even better on top in such kind of skirmish.

This morning I really was motivated to build parametric assembly on top of Assembly3. 
Was hoping that "Lock" constraints might resemble a parent part, and "Attachment" used to 6DOF match LCS.
... until I really dared to test ... that's pre-alpha - both in 0.21.2 and in nightly dev. Not even the simplest own tutorials work.
https://forum.freecad.org/viewtopic.php?t=86838

I sincerely wish "them" success for the huge endeavour to carry this concept to maturity.
But I'll neither wait nor spend my lifetime for that.

In terms of FOSS, I feel the point for a major fork.
I don't think I can spend the time for indepth development.
But I might contribute in testing and documentation.

So if time comes, let's keep contact.
Till then, I'll build my own projects on spreadsheets aggregating transformations, avoiding any assembly framwork at all, as long as possible.

Zolko-123, 2024-06-07:
This can be closed I think

closed 2024-06-07 by Zolko-123