Zolko-123/FreeCAD_Assembly4 #478: Workbench failure at startup

neildarlow, 2024-03-19:
Just updated to version 0.50.10 and Assembly4 reports an error at startup.

Report Window content:
```
18:00:55  Initializing Assembly4 workbench (0.50.10) .............invalid syntax (makeArrayCmd.py, line 213)
18:00:55  Traceback (most recent call last):
  File "<string>", line 154, in Initialize
  File "/app/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

FreeCAD version:
```
[code]
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Help 1.0.3
  * Behave-Dark-Colors 0.1.1
  * Assembly4 0.50.10
  * fasteners 0.5.14
  * lattice2 1.0.0
[/code]
```

Zolko-123, 2024-03-19:
Can you please update to 0.50.11 and try again ?

neildarlow, 2024-03-19:
Hi,

It was version 0.50.12 by the time I updated and that looks good. Thank
you.

Regards,
Neil Darlow
> Message ID: <Zolko-
> ***@***.***>

closed 2024-06-07 by Zolko-123