Zolko-123/FreeCAD_Assembly4 #484: I cannot export assembly4 'linear array' Using FreeCAD without GUI

3xxx, 2024-03-21:
Hi zolko!
I made the model using the linear array of A4 and export the model in the python code of [headless mode](https://wiki.freecad.org/Embedding_FreeCAD )and only got the first part, not the arrayed part.
some python code like this:
```python
......
doc = FreeCAD.open(inputfile)
.....
ss = doc.getObject('Spreadsheet')
......
ss.set('E'+str(index+2), str(parameter[index]))
......
FreeCAD.ActiveDocument.recompute()

Import.export(__objs__, outputfile)
```
Does linear array not support headless mode?
Thanks you!

Zolko-123, 2024-03-22:
I have never tried

3xxx, 2024-03-23:
I solved it by running python code in freecadcmd, thanks you.

Zolko-123, 2024-06-03:
if it's solved, can you close the issue ?

closed 2024-06-07 by Zolko-123