Zolko-123/FreeCAD_Assembly4 #491: name 'QtCore' is not defined

snelweg, 2024-05-05:
When switching over to the ASM4 WB:

```
13:20:11  name 'QtCore' is not defined
13:20:11  Traceback (most recent call last):
  File "/home/ton/.local/share/FreeCAD/Mod/Assembly4/./InitGui.py", line 100, in Initialize
    msgBox.setWindowFlags( QtCore.Qt.WindowStaysOnTopHint )
```

Has been an error for the past 4 appimages. What is wrong and how can I tfix it?

`OS: Linux Mint 21.3 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37100 (Git) AppImage
Build type: Release
Branch: main
Hash: ede6c811fd39d68a44949601f2eb7ecc8a67fdb7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * Render 2024.1.23
`

Vincenzoferrara, 2024-05-17:
I solved it by adding to line 100 In the file InitGui.py

`from PySide import QtCore`

Zolko-123, 2024-06-07:
Should be solved in the latest dev branch

closed 2024-06-07 by Zolko-123