Zolko-123/FreeCAD_Assembly4 #493: Since you were banned from Freecad forum, you could open a discord forum. What do you think?

leoheck, 2024-05-31:
Hey, since you were banned from Freecad forum, you could open a Discord forum. What do you think?
If you do so, let me know and I will post this on the Freecad forum for the users that follow Assembly 4 instructions to find it.

Zolko-123, 2024-06-03:
I added a [Discussion](https://github.com/Zolko-123/FreeCAD_Assembly4/discussions) tab for this purpose, though I don't know how to use it. May-be can you give it a try ?

I do have Discord but I only use it to communicate with my sport club.  I also have Telegram, but I don't think they're well adapted for such development purposes.

Zolko-123, 2024-06-03:
If you're willing to post a link on the forum so others can have the info would also be nice

leoheck, 2024-06-04:
Yes, I will do that. That's a good starting point. Also, that forum is horrible to use. so win-win.

leoheck, 2024-06-04:
Cool, I am going to close this issue since it is all addressed now. Thanks.
[image]

closed 2024-06-04 by leoheck