Zolko-123/FreeCAD_Assembly4 #496: Command FSChangeParams changed to Fasteners_ChangeParameters

FloatingPoint75, 2024-06-05:
Hello,
when launching the "Change Fasteners Parameters" an error occurs, as the command is not found. Replacing all instances of "FSChangeParams" with the actual command name "Fasteners_ChangeParameters" in FastenersLib.py solved the issue.

Please update the file in the repo.

Thanks

Zolko-123, 2024-06-07:
Why do people change keywords randomly ? Geeeez, it's more work for no benefit

Zolko-123, 2024-06-07:
done in latest dev branch

Zolko-123, 2024-06-07:
Done in the latest release: can you please update and test ?

FloatingPoint75, 2024-06-07:
> Done in the latest release: can you please update and test ?

Tested, it works, but I get this error:
18:40:13  Traceback (most recent call last):
  File "C:\Users\giuli\AppData\Roaming\FreeCAD\Mod\fasteners\.\FSChangeParams.py", line 403, in getStandardButtons
    return QtGui.QDialogButtonBox.Ok + QtGui.QDialogButtonBox.Cancel
<class 'TypeError'>: unsupported operand type(s) for +: 'PySide2.QtWidgets.QDialogButtonBox.StandardButton' and 'PySide2.QtWidgets.QDialogButtonBox.StandardButton'

Zolko-123, 2024-06-07:
yes I see it here too. That should be a | and not a + .  You can change it in your local Mod directory

It's a bug in the fasteners workbench, happens also within the fasteners WB itself, could you please report it to them ?

FloatingPoint75, 2024-06-08:
Thanks for the quick fix! I opened the issue on the Fasteners WB repo as well.

Have a great weekend!

FloatingPoint75, 2024-06-08:
He fixed it :)

Zolko-123, 2024-07-05:
> He fixed it :)

if it is fixed can you please close the issue ?

FloatingPoint75, 2024-07-05:
Sure, thanks for your support!

closed 2024-07-05 by FloatingPoint75