Zolko-123/FreeCAD_Assembly4 #503: Variant Part variables not persistent or saving bug

OrdoScintilla, 2024-06-26:
Hi. Thanks Zolko for your time and work.

I can create a "source part" in file 1 and retrieve all the features (sketch constraints, pad, fillet, chamfer, etc.) of my part from "Variables".

Once I create the "destination assembly" in file 2, I can insert 2 or more variants successfully. Then I can edit the variables for each variant successfully. No problems.

I don't know why, but at first I could see immediately how the part updated in the 3D model each time I edit a variable. Now I have to right-click on the Part that is inside the Variant_Part and select "recompute object" for changes to take effect. This is no big issue, It's just weird.

The big problem:
Once I edited all the variables for each variant, and save my file. There appears an error message (inserted below). If I then "close all" and reopen my files, the variants and their variables get reverted to the variables inside my "source part" file (file 1). Thus, it seems that the error is aborting all the "save".

I have only been trying variants for 1-2 days and have created over 20 couples of files as source and destination to simplify the model and see if there is another issue. At first I was even using Spreadsheets and "Configuration Table" with multiple features. As issues arose I simplified everything to just a square profile with minimum constraints and just 3 variables (base, height, length). Nonetheless, error persists.

Then I thought, maybe my copy of WB is corrupt so I updated it. No luck,

Finally, before posting here, I even uninstalled my whole FreeCAD with all my personalized settings and workbenches. Reinstalled a clean FreeCAD (v0.21.2) with only "Assembly4" workbench added (v0.50.14).

I'll give a simplified run-through of what I do:

1.- Create file 1 and save it as "source".
2.- In "Assembly 4" workbench, create a New Assembly, then rename it to "Source_Assembly"
3.- Inside it create a "New Part", rename it to "Src_Part".
4.- Inside it create a "New Body", rename it to "Src_Body".
5.- Inside it create a "New Sketch", by default name "Sketch_1".
Add a rectangle.
Fully constrain it with current size.
6.- In "Part Design" workbench add a "pad" with default length.
7.- In "Assembly 4" select "add variable"
Type: float
Name: "Base"
Value: 10
8.- Do the same for variable "Length" and "Height"
9.- Go back to sketch and edit all constraints to "Variables.Base", "Variables.Height" using the expression editor (formula editor opens up).
10.- Edit the pad (right click Pad, select "Edit Pad") dimension length to "Variables.Length" using the expression editor (formula editor opens up).
11.- Go bak to "Variables" container and edit each one to make sure constraints work ok.
12.- Save and close. Everything's ok.

Second file:
13.- Create file 1 and save it as "destination".
14.- In "Assembly 4" workbench, create a New Assembly, then rename it to "Dest_Assembly"
15.- Save and close. Everything's ok.

Merge:
16.- Open both files.
17.- In "destination" click on "Dest_assembly"
18.- In "Assembly 4" workbench, "create a Variant Part",
Select Part: Source#Source_Assembly
Name for the link: Source_Assembly_var
clink Insert
In "Tasks" in combo "Attach to" select "Parent Assembly"
In "Select LCS in Parent" select LCS_Origin
Change nothing else, just OK.

Inside "Dest_assembly" a Variant Part called "Source_Assembly_var" is created.
19. Under "Source_Assembly_var" select "Variables" container and edit the parameters (for instance all to "30").
20.- Under "Source_Assembly_var" select "Src_Part" and right-click "Recompute Object". Everything's ok.}

21.- Repeat steps 17-20, changing:
in 18.-
Name for the link: Source_Assembly_var_2
In "Tasks" Under "X translation" type "100"
in 19.-
Under "Source_Assembly_var_2" select "Variables" container and edit the parameters (for instance all to "50").
in 20.-
Under "Source_Assembly_var_2" select "Src_Part" and right-click "Recompute Object". Everything's ok.

Finally:
22.- "Save All"

The following Error Message shows up:

`11:20:15  PropertyPythonObject::toString(): failed for <class 'Asm4_objects.VariantLink'>
11:20:15  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
11:20:15  PropertyPythonObject::toString(): failed for <class 'Asm4_objects.VariantLink'>
11:20:15  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
11:20:15  PropertyPythonObject::toString(): failed for <class 'Asm4_objects.ViewProviderVariant'>
11:20:15  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderLink is not JSON serializable
11:20:15  PropertyPythonObject::toString(): failed for <class 'Asm4_objects.ViewProviderVariant'>
11:20:15  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "C:\Program Files\FreeCAD\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderLink is not JSON serializable`

Step 23.- Close All and reopen.

On reopen, the following error shows up, which was already brought up.

`Exception opening file: C:/Users/../AppData/Local/Temp/FreeCAD/Cache/FreeCAD_Doc_2..0 [File '..' does not exist!]`

Result:
Everything is reverted to the variables as set in "Source File". No Variant is saved with its variations.

Please could you advise? I would really love for this functionality to work for me.

Thanks again for all the time invested.

Zolko-123, 2024-06-27:
Could you please try with FreeCAD version 0.21.1 ? (sorry I don't have the time right now)

0.21.2 introduced a regression caused by some bleeding-edge Python 3.11 serialisation issue, that could provoke this error. I recommend using 0.21.1 (that's what I'm using).

Zolko-123, 2024-06-28:
I just tested it with 0.21.1 and 0.21.2 and it works perfectly with .1 and gives that error in .2 . (you can simplify your example to a single Part and a single variable and importing it as VariantLink into an assembly)

This is a regression in FreeCAD, badly "solving" a bleeding-edge issue with bleeding-edge 3.11 Python, backported for no reason to the stable branch by incompetent maintainers, and fortunately fixed later by **realthunder** in his branch and picked for the 0.22-dev branch. It works in all FreeCAD versions except the "stable" 0.21.2 . There is nothing that Assembly4 can do.

See:
https://github.com/FreeCAD/FreeCAD/issues/10460
https://github.com/FreeCAD/FreeCAD/commit/fbe2fef66cfcee5685d435db57cf34a364315ae9
https://github.com/realthunder/FreeCAD/commit/9a7e85e917c102161e92a7758fd801ea0aaa4587

I recommend using FreeCAD 0.21.1

OrdoScintilla, 2024-06-30:
Thanks.
Went back to 0.21.1 and tried it out. I'm still getting bugs. 

The variant part adequately changes variables and updates with the set variables.
The variant part also changes a Spreadsheet configuration table that selects between different profiles I made.

Still need to "Recompute Object" to see changes in both cases. This has to be done at the "Source Part" inside the "Variant Part". It all seems good here.

But if I "Recompute Object" at the "Variant Part" level, then everything gets reverted to the variables that were set in the source file.

This is also true if I select "Solve and update assembly". But in this case, all the variants parts are reverted.

Also parts are reverted to source variables when I save file and close and reopen.

These are sometimes preserved, but there is no clear pattern in what I do to preserve or revert.

Would you appreciate me sending a video of my process?

Zolko-123, 2024-06-30:
> 19. Under "Source_Assembly_var" select "Variables" container and edit the parameters (for instance all to "30")
> 20. Under "Source_Assembly_var" select "Src_Part" and right-click "Recompute Object". Everything's ok.}

wait ... you should change the variant parameters in the destination ! There is no "Variables" container in the _variantLink_

[image]

Zolko-123, 2024-06-30:
ah I see now ... no, it doesn't work that way, the variant variables need to be stored in the assembly, so it's not enough to edit the Variables in the _variantLink_. 

Good catch though

OrdoScintilla, 2024-07-01:
> > 19. Under "Source_Assembly_var" select "Variables" container and edit the parameters (for instance all to "30")
> > 20. Under "Source_Assembly_var" select "Src_Part" and right-click "Recompute Object". Everything's ok.}
> 
> wait ... you should change the variant parameters in the destination !

I'm doing it in the destination. In my explanation, you see from step 13 onwards, everything is in the second file.

The following screenshots are both in the second file (the destination):

The "Variables" container for the Main Assembly "Dest Assembly" in the second file has no variables to be changed.
[image]

The "Variables" container for the Variant which is under "Dest Assembly", and has its own name as "Source_Assembly_Var" (because this was linked from the origin file) has the variables Base, Height, Length. These are the variables that change the volume of the part. Which is where I've been editing, but these do not persist after closing.
[image]

I tried both in FreeCAD 0.21.1 and now I'm with 0.22.0 dev and the result is the same.

With regards to your comment:
> > ah I see now ... no, it doesn't work that way, the variant variables need to be stored in the assembly, so it's not enough to edit the Variables in the variantLink.

Should I be creating the same variables I'm using for constraints at this level? But that would immediately change all the parameters for all variants.

Zolko-123, 2024-07-01:
what do you see in the properties when you select _Source_Assembly_var_ ?

OrdoScintilla, 2024-07-01:
Aaaaah. Now I get it. There are "Variant Variables" inside _var too.  

Sorry for the time-consuming, innecesary questions.

It works and now it updates without recomputing.

I hope this serves as an example for many other people.

Zolko-123, 2024-07-01:
It will serve as an example to provide documentation. This is one of my favorite features but is not very well explained, sorry for that

OrdoScintilla, 2024-07-02:
Ok, I found it exactly to be what I need. The issue of explanation I will try to do here for future inquirers. So this issue will be closed. There is another issue I'll bring up in another branch, which has to do with LCS constraints.

closed 2024-07-02 by OrdoScintilla