Zolko-123/FreeCAD_Assembly4 #515: Could the Asm4 message be removed from the Report View when initializing through the GUI?

leoheck, 2024-09-12:
Hi @Zolko-123 . Can we have the initialization messages removed, so the Report view starts clean after launching FreeCAD with the GUI?

For instance, the new `1.0-RC1` does not show any red or any messages at all after launching. The Report view starts clean.
See, only Asm4 shows text.

[image]

However, when launching on the terminal, it is fine to show the messages.
See the `Sheet metal` in the following image, for instance. This message appears only when FreeCAD is launched from the terminal, but not when it is launched on GUI.

[image]

Zolko-123, 2024-09-13:
actually ... this message was nice before, because one could see the dots advancing while the WB was loaded. Unfortunately, the console was changed so the dots don't appear progressively but all at once at the end. So it's not very user-friendly anymore. 

Let me think about something

leoheck, 2024-09-13:
hm, at the command line it works like that.
it personally think it would be nice to have a clean console when starting Freecad.

closed 2025-02-21 by leoheck