Zolko-123/FreeCAD_Assembly4 #520: Changing a fastener's parameter changes all the labels of all the fasteners added

Sil68, 2024-10-04:
# Current situation
I have got an assembly comprising of various parts. I am using the fastener workbench to add some fasteners to this assembly.

# Observation
Whenever I am change a parameter of any of the fasteners, all labels of all fasteners are getting automatically changed adhering to some default sequence labelling scheme.

# Expectation
Each label of each fastener added shall remain as-is. In particular when having assigned a custom label to a fastener.

Zolko-123, 2024-10-05:
That is strange : did you see this before ? What versions (FreeCAD, Asm4, Fasteners...) are you using ?

Zolko-123, 2025-01-30:
Is this still relevant ?

Sil68, 2025-01-30:
Last time I checked, it still was

Zolko-123, 2025-01-30:
strange, it works nominally for me. 
could you please provide the answers to these questions so I can investigate : 

> What versions (FreeCAD, Asm4, Fasteners...) are you using ?
> Also: do you use the Asm4 interfaces to add and change fasteners, or do you use the native Fasteners tools ?

Sil68, 2025-01-30:
Failed to reproduce the observed behaviour.

So I reckon the issue is fxed.

I’m using FreeCAD

OS: macOS 15.3
Architecture: arm64
Version: 1.1.0dev.39980 (Git) Conda
Build type: Release
Branch: main
Hash: 836ad91f3ddd3ec63b84b8e3f87a767d5465261d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * MeshRemodel 1.10.36
  * ThreadProfile 1.92.0
  * Reporting
  * btl 0.9.9
  * BillOfMaterials 1.0.2
  * freecad.gears 1.3.0
  * woodworking 0.21.2.33771.0.23.1
  * OpenTheme 2024.9.1
  * ArchTextures
  * Defeaturing 1.2.3
  * Gridfinity 0.8.0
  * FEMbyGEN 2.3.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.36
  * Lithophane
  * lattice2 1.0.0
  * Estimate 0.1.2
  * Design456 0.0.1
  * WB_Organizer 2024.1.29
  * sheetmetal 0.7.5
  * FEM_FrontISTR 0.2.0
  * 3D_Printing_Tools
  * Curves 0.6.54
  * Assembly4 0.50.15

The assembly is done facilitating the fastener wb.

Zolko-123, 2025-01-30:
Thanks, I'm closing this then

closed 2025-01-30 by Zolko-123