Zolko-123/FreeCAD_Assembly4 #534: Error creating new object (dev branch)

leoheck, 2025-02-15:
I am creating a Body with the Asm4, I am seeing this.

```
22:50:40  Running the Python command 'Asm4_newBody' failed:
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/Assembly4/./newPartCmd.py", line 91, in Activated
    App.ActiveDocument.getObject('YZ_Plane'+index).ViewObject.ShapeColor=(1.0, 0.0, 0.0)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

'NoneType' object has no attribute 'ViewObject'
```

Zolko-123, 2025-02-15:
What FreeCAD version are you using ? I think "they" changed the name of the Origin objects sometimes

Zolko-123, 2025-02-15:
This is a ridiculous regression : when creating a Part or Body, it creates the `Origin` object, containing the axes and planes, and creates also a point called also `Origin` ! Which is therefore renamed `Origin001` since 2 objects cannot have the same name. When you then create a new Part or Body, the `Origin` object is renamed `Origin002` (since `Origin` and `Origin001` already exist), while the axes and planes are renamed `X_Axis001` since `X_Axis` already exists. Therefore, the index that is calculated from `Origin002` is wrong.

The error is clearly on FreCAD side to create 2 objects with a known and existing name twice : `Origin` as the group of axes and planes, and `Origin` as a point. This is a bug and regression in the FreeCAD development branch, please report it there.

leoheck, 2025-02-16:
> What FreeCAD version are you using? 

Hm. I always use the latest weekly, which is currently `1.1.0, Libs: 1.1.0devR40253`.

> This is a bug and regression in the FreeCAD development branch, please report it there.

Alright, I can do that.

leoheck, 2025-02-16:
The 1.1-dev has this new format of the origin elements.

[image]

The one on the left is the ASM4 (created on 1.0, without any errors)
The one on the right was created in 1.1-dev

Zolko-123, 2025-02-16:
Yeah I've seen that. But that is not even the problem : the problem is that it creates an additional element, a point/vertex, that is called "Origin" like the "Origin" coordinate system. So they create at the same time 2 objects with the same name, that is a bug.

leoheck, 2025-02-16:
Hm, these objects have the same Label, not the same Name.
There is this `Origin` with the label `Origin` and inside the point/vertex is `Origin001` labeled `Origin`
Maybe Asm4 is using Label instead of Name.
I also have on my Freecad a setting enabled to alow the use of objects with the same Labels.

leoheck, 2025-02-16:
The problem should be something different. 
There is only one App::Origin per Body.

```python
# using the existing code on Asm4
for origin in App.ActiveDocument.findObjects(Type='App::Origin'):
  parent = origin.getParentGeoFeatureGroup()
  print("> Body_Name:", parent.Name, " | Body_Label: ", parent.Label, " | Origin_Name:", origin.Name, "Origin_Label:", origin.Label)

# Output  
# > Body_Name: Body    | Body_Label:  Body    | Origin_Name: Origin    Origin_Label: Origin
# > Body_Name: Body_2  | Body_Label:  Body_2  | Origin_Name: Origin002 Origin_Label: Origin
```

Zolko-123, 2025-02-16:
They are **created** with the same name `Origin`, but since that's not possible FreeCAD renames the second one to `Origin001`.

I just checked : when you import a STEP assembly with nested parts, the names of the Origin and X_Axis and so-on are all scrambled. Before that, they all had the same index, now they all have inconsistent names : the Origin group is called `Origin014`, the Origin point is called `Origin015`, and the axes and planes inside are called `X_Axis007`.

The naming itself is inconsistent : the axes are called `X_Axes` ..., the planes `XY_Plane`..., so the origin point should be called `O_Point` or similar 

 This was done by someone who doesn't know FreeCAD's internals, it's clearly a bug. 

My advice : use the 0.21 branch, the 1.x are evolutionary dead-ends, they will never be any good for assemblies, because the development process itself is broken.

leoheck, 2025-02-16:
The first Body is created fine, the second that has this issue.
[image]Thr

Zolko-123, 2025-02-16:
> There is only one App::Origin per Body

the other `Origin` is an `App::Point`

> The first Body is created fine, the second that has this issue

because the Origin point is created after the axes and planes, therefore the naming is consistent for the first container

leoheck, 2025-02-16:
The Asm4_NewBody is repeating things for all objects.
If I use it to create other objects, it is doing things over and over for all the existing objects.
Is this intended?

[image]

leoheck, 2025-02-16:
and it breaks on the last one only (the one that is currently being created.

Zolko-123, 2025-02-16:
Don't worry, Werner will correct that bug

leoheck, 2025-02-16:
Here, https://github.com/Zolko-123/FreeCAD_Assembly4/pull/535

Zolko-123, 2025-02-16:
> It was redoing things for all existing parts. It should customize only the part being created.

you're right, that's bad. I'll check your fix, thank-you (it's still a bug to create 2 objects with the same name, but I don't care anymore about the crap that they do)

leoheck, 2025-02-16:
Nice, hope my fix is good enough for the Asm4. Otherwise, let me know and I will adjust it.
I started to create objects with Asm4 this week since I can create objects with names that make sense instead of just with meaningful labels.

Zolko-123, 2025-02-16:
some minor errors but is good, thank-you. Please update and check

leoheck, 2025-02-16:
It is working just fine, thanks.

One extra thing, can we have the Origin visibility hidden by default?
If you are ok with this, I can make a PR.

The PartDesign_Body (the last `Body` in the following picture) is created with the Origin hidden.

[image]

Zolko-123, 2025-02-17:
> can we have the Origin visibility hidden by default ?
> The PartDesign_Body (the last `Body` in the following picture) is created with the Origin hidden.

why is this good ?

leoheck, 2025-02-17:
I would suggest to have the origin and the lcs_origin hidden there is no reason for these things to be visible. they are just references that are always at the same place for each body. but since lcs_origin is visible the origin is not necessary to be visible and it will also be like in the PartDesing body. 

Also, thinking deeper here, Asm4 should allow to select object's origin as the lcs_origin so this lcs_origin would not need to be added by default. Only if the user wants to set it in a different place.

Zolko-123, 2025-02-17:
I like the way it is and it works. If more people complain we can we can re-evaluate this again

closed 2025-02-17 by leoheck