Zolko-123/FreeCAD_Assembly4 #539: Issue on 1.1.0-dev R40408

leoheck, 2025-02-26:
Now something really bad happened.

[image]

```
Running the Python command 'Asm4_placeLink' failed:
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/Assembly4/./placeLinkCmd.py", line 81, in Activated
    ui = placeLinkUI()
         ^^^^^^^^^^^^^
  File "/home/lheck/.local/share/FreeCAD/Mod/Assembly4/./placeLinkUI.py", line 107, in __init__
    self.drawUI()
  File "/home/lheck/.local/share/FreeCAD/Mod/Assembly4/./placeLinkUI.py", line 922, in drawUI
    PathGuiUtil.QuantitySpinBox(self.XtranslSpinBox, self.selectedObj, "Placement.Base.x")
  File "/tmp/.mount_FreeCAJCOAdl/usr/Mod/CAM/Path/Base/Gui/Util.py", line 129, in __init__
    self.lastWidgetText = self.widget.text()
                          ^^^^^^^^^^^^^^^^

'PySide2.QtWidgets.QWidget' object has no attribute 'text'
```

leoheck, 2025-02-26:
It looks like this is happening on version `v0.50.18-dev` but not on `v0.50.17`

closed 2025-02-27 by leoheck

Zolko-123, 2025-02-27:
I think we'll go to 0.60 in the next release. I want to include the translation code, and move to code to a sub-directory to clean up the main repo directory