Zolko-123/FreeCAD_Assembly4 #543: Ignore/Cancel buttons do nothing when Inserting Links.

leoheck, 2025-03-01:
It seems that Ignore and Cancel are not doing anything.
Can you tell me what they should do?

[image]

I would like to insert a part, then Ignore or Cancel to abort the operation, but both Ignore and Cancel are keeping the link.

Zolko-123, 2025-03-06:
yes, that panel is only for the placement/attachment of the link. But the link is created before that. I don't see a way of changing this (may-be some better explanation, but the link must exist during the placement task)

leoheck, 2025-03-06:
Hm, I mean...

I would like to be able to cancel the operation without keeping the link placed. 
If I click Cancel or Ignore, the link is still placed. 
Which one (Ignore or Cancel) should do that?
If I am not mistaken the placeLink is used to place a new Link and also to edit an existing placement.

We could have a single button, that is... Cancel. 
- If the placement is being edited, then it should cancel the current operation.
- IF the link is being placed at that time, the cancel should remove the Link from the ASsembly.

Do you agree?

Zolko-123, 2025-03-06:
It's more complicated than that : 

When editing the placement of a link, **Cancel** cancels the new placement and restores the previous one. **Ignore** removes Asm4 placement propertied leaving the link in manual placement (so another solver can take over : see I thought about such things very much in advance, though I didn't have time or competence to implement it).

When this dialog is called after inserting a new link, the **Cancel** and **Ignore** buttons do the same thing.

In order to use **Cancel** as a way to cancel the insertion of a new link itself, this dialog should somehow "know" that it's a new virgin link. But it should not delete an old link that was placed manually. A use-case for example is to use the dialog to place a furniture on a floor, and then "liberate" it from any constraint by using **Ignore**, and move the furniture by hand in the horizontal plane. This **Cancel** button must respect that use-case