Zolko-123/FreeCAD_Assembly4 #547: Fasteners and cyclic dependencies.

leoheck, 2025-03-05:
Using fastners with Asm4 I cannot place them inside the Assembly, freecad complains with cycling dependencies.

Do you know someone about this issue?

Zolko-123, 2025-03-05:
Is this new ? It always worked. What versions are you using ?

leoheck, 2025-03-05:
I am not sure, maybe.

When I create screws with the Fasteners WB they are placed outside of the Assembly, when I drag them inside I see this message about cyclic dependencies. 

If I use Asm4 to create the screws, it also crestes out of teh Assembly. Dragging them inside Freecad still complains. 

Actually dragging anything (any links inside the Assembly I see the same message).

I will investigate this better and create a sample file.

leoheck, 2025-03-05:
When I add a screw, it is added out of Assembly if a hole is selected, like this:
And even if a hole was selected it does not change the size to fit the hole. I am not sure if this was expected, maybe not..

[image]

Draggin it inside works and there are no messages from FreeCad

[image]

Then using Fasterners tools, I can move it to the right place, and change its size to match the hole.

[image]

But then it complains about this:
It would be nice to be able to use this tool so I would not need to make extra LCS.
But I am not sure if Asm4 supports it.
For now, using LCS is a better solution since it works and I could do what I wanted to do.

[image]

-----

If I add a screw without selecting anything, then Asm4 creates the screw in the Assembly - this could be my mistake.

-----

If I create the screw with the Fasteners, it places it outside of the Assembly, but if I drag it inside, Freecad places it there but complains with multiple messages like this.

[image]

---- 

Ok, I could find a solution that works, which is

- Create the fastener with Asm4 tool (only), without selecting anything.
- Place it with an LCS (only)
- Then I can use tools to make patterns if needed

[image]

The sample file is here.

[cyclic_dep.FCStd.zip](https://github.com/user-attachments/files/19094995/cyclic_dep.FCStd.zip)

Zolko-123, 2025-03-06:
>     * Create the fastener with Asm4 tool (only), without selecting anything.
>     * Place it with an LCS (only)
>     * Then I can use tools to make patterns if needed

That's the intended usage. Although I mostly use "Hole Axis" for the placement of fasteners. I'm trying to implement a faster workflow:

* select a circle edge
* click "Insert Fastener" which will create a "Hole Axis" datum, will create a fastener, and attach the fastener to the newly created datum axis
* ideally, the fastener created should be of size of the circle
* the type must still be adapted

leoheck, 2025-03-06:
Oh yeah, this workflow would be awesome.
Would you like some help with that?
Also, the latest bullet is not a problem if it is possible to modify multiple fasteners simultaneously.

leoheck, 2025-03-06:
I dont have some time, but I will make some to check this haha.

leoheck, 2025-03-06:
`Asm4_shapeBinder` is creating these objects with a strange name, I am not sure if this is what you want.

[image]

And when passing the mouse over this object, or clicking on it, I get these messages:

```
<QuickMeasure> QuickMeasure.cpp(97): 0x5c040bd0dea0 : Standard_TypeMismatch: TopoDS::Edge
<QuickMeasure> QuickMeasure.cpp(97): 0x5c040bd0dea0 : Standard_TypeMismatch: TopoDS::Edge
<QuickMeasure> QuickMeasure.cpp(97): 0x5c040bd0dea0 : Standard_TypeMismatch: TopoDS::Edge
<QuickMeasure> QuickMeasure.cpp(97): 0x5c040bd0dea0 : Standard_TypeMismatch: TopoDS::Edge
<QuickMeasure> QuickMeasure.cpp(97): 0x5c040bd0dea0 : Standard_TypeMismatch: TopoDS::Edge
<QuickMeasure> QuickMeasure.cpp(97): 0x5c040bd0dea0 : Standard_TypeMismatch: TopoDS::Edge
```

This using `1.1-dev`

leoheck, 2025-03-06:
The `ShapeBinder` was removed from the toolbar of PartDesign, only the `SubShapeBinder` is there.

[image]

But the function still exists in the menu `Part Design > Create a shape binder`

leoheck, 2025-03-06:
Creating a `SubShapeBinder` (green) creates a cyclic dependency, but creating a `ShapeBinder` (blue) with the Assembly is Visible since it is created inside the linked Body from the Assembly.

![Image](https://github.com/user-attachments/assets/3eca8e86-babb-45b9-ac44-37dbb381a03d

Now if the Body is Active. It creates the `SubShapeBinder` fine.

leoheck, 2025-03-06:
Ah, if the Assembly is active, I can drag the Screw created with the Fasteners WB inside it without having this Cyclic reference. So the Fastener ends up inside the Assembly.

leoheck, 2025-03-06:
But once I place it using Fasterner WB tools, it complains about the cyclic dependency again.

leoheck, 2025-03-06:
When I created a Hole_Axis freecad crashed.
I hate it so much when it crashes.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x740eb6e45330]
#1  0x740e90d51fc6 in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /tmp/.mount_FreeCAgCJaKj/usr/lib/./libTKBRep.so.7.7+0x26
#2  0x740e90d63b21 in BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) from /tmp/.mount_FreeCAgCJaKj/usr/lib/./libTKBRep.so.7.7+0xd1
#3  0x740e90d64d35 in BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) from /tmp/.mount_FreeCAgCJaKj/usr/lib/./libTKBRep.so.7.7+0xd5
#4  0x740e9c3cf67e in Measure::Measurement::length() const from /tmp/.mount_FreeCAgCJaKj/usr/lib/Measure.so+0x15e
#5  0x740e874aac0e in MeasureGui::QuickMeasure::printResult() from /tmp/.mount_FreeCAgCJaKj/usr/lib/MeasureGui.so+0x34e
#6  0x740e874ab8ed in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCAgCJaKj/usr/lib/MeasureGui.so+0x5d
#7  /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x740eb74eb6f2]
#8  0x740eb74ef34a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Core.so.5+0x2a
#9  0x740eb74e064d in QObject::event(QEvent*) from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Core.so.5+0x7d
#10  0x740eb83703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#11  0x740eba17f629 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libFreeCADGui.so+0xc9
#12  0x740eb74b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Core.so.5+0x108
#13  0x740eb7509f51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Core.so.5+0x381
#14  /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Core.so.5(+0x30a75c) [0x740eb750a75c]
#15  /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x740eb5b10ad7]
#16  /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x740eb5b13ac7]
#17  /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x740eb5b140ce]
#18  0x740eb750aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Core.so.5+0x66
#19  0x740eb74b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Core.so.5+0x11b
#20  0x740eb74bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libQt5Core.so.5+0x82
#21  0x740eba0decad in Gui::Application::runApplication() from /tmp/.mount_FreeCAgCJaKj/usr/bin/../lib/libFreeCADGui.so+0x70d
#22  /tmp/.mount_FreeCAgCJaKj/usr/bin/freecad(+0x7906) [0x5c03dbba3906]
#23  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x740eb6e2a1ca]
#24  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x740eb6e2a28b]
#25  /tmp/.mount_FreeCAgCJaKj/usr/bin/freecad(+0x7c8e) [0x5c03dbba3c8e]
```

Zolko-123, 2025-03-06:
> When I created a Hole_Axis freecad crashed.
> I hate it so much when it crashes.

yeah I just tried and it crashed indeed. 0.21 never crashes. I don't understand how you support that crap, and why

leoheck, 2025-03-07:
Because it is progress. While assembly is important, improving how objects are created is important as well. And version 1.1 is in development they are not saying it is stable. So while some can live in the past using stable things, others can live in the future having access to new features but knowing things may not be as stable as we would like and because of that tje development of the next version can move faster.

Anyway, an issue was placed for Freecad devs, another user already confirmed the thing. Let's see what they respond.

https://github.com/FreeCAD/FreeCAD/issues/20035

Zolko-123, 2025-03-07:
Thanx

leoheck, 2025-03-07:
I am starting to think that a good alternative approach for screws is to have them as Parts.