Zolko-123/FreeCAD_Assembly4 #550: Objects generated using Assembly4's image function cannot be exported to GLB, STP, or IFC

3xxx, 2025-03-13:
assembly0.50.15
Objects generated using Assembly4's image function cannot be exported to GLB, STP, or IFC. Whether it is the Freecad link version or the official version, the version information is as follows
[assemblytestmirror.zip](https://github.com/user-attachments/files/19222397/assemblytestmirror.zip)

link:
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.8, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods: 
  * A2plus 0.4.66
  * A2plus.backup1724687131.225395 0.4.64 (Disabled)
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * Assembly4.backup1724683001.0928686 0.50.9 (Disabled)
  * BillOfMaterials 0.0.7
  * Help 1.0.3
[/code]

official:
OS: Windows 10 build 19044
Architecture: x86_64
Version: 1.1.0dev.39980 (Git) Conda
Build type: Release
Branch: main
Hash: 836ad91f3ddd3ec63b84b8e3f87a767d5465261d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.66
  * A2plus.backup1724687131.225395 0.4.64 (Disabled)
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * Assembly4.backup1724683001.0928686 0.50.9 (Disabled)
  * BillOfMaterials 0.0.7
  * Help 1.0.3

Zolko-123, 2025-05-19:
You're using realthunder's Link branch ! On my 0.21.3 version it works as expected : 

[image]

[image]

```
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.3.33883 (Git)
Build type: Unknown
Branch: development
Hash: b3f54a5c52989db3f71ea8051ed381f90551d20e
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
  * A2plus 0.4.56a
  * CurvedShapes 1.0.3
  * fcgear 1.0.0
  * sheetmetal 0.2.50
  * TabBar
  * Render 2022.2.0
  * Curves 0.5.1
  * fasteners 0.5.21
  * Assembly4_dev 0.50.19-dev
```

I'm afraid I can't help there

3xxx, 2025-05-23:
Thank you.
 I test it in 1.1.0dev,when export to stp,it like this:the origin body is just line.
[image]
[image]

when export to glb,the mirror body is nothing.

3xxx, 2025-05-28:
Thank you，i‘ll try 0.21