Zolko-123/FreeCAD_Assembly4 #551: fc crashing when selecting two asm4 lcs' in treeview

Sil68, 2025-03-14:
# Setting
:: I have created a simple box in part design wb;
:: switched to asm4 wb;
:: created several lcs' with asm4.

# Problem
:: selecting just one of aforementioned lcs' in treeview works perfectly fine;
:: selecting more than two lcs' works also as intended;
:: only when selecting exaclty two lcs' fc is reproducibly crashing.

# fc
OS: macOS 15.4
Architecture: arm64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/108
Installed mods: 
  * MeshRemodel 1.10.37
  * ThreadProfile 1.93.0
  * Reporting
  * Launcher
  * 3DfindIT 1.2.0
  * BillOfMaterials 1.0.2
  * woodworking 0.21.2.33771.0.24.1
  * ThreadProfile.backup1741713547.9851232 1.92.0 (Disabled)
  * FreeGrid 2.2.0
  * Defeaturing 1.2.3
  * Gridfinity 0.10.4
  * FEMbyGEN 2.3.0
  * fasteners 0.5.37
  * Lithophane
  * Assembly4.zip
  * lattice2 1.0.0
  * sheetmetal 0.7.21
  * 3D_Printing_Tools
  * FrameForge 0.1.3
  * Assembly4 0.50.18

[fc_asm4_lcs.FCStd.zip](https://github.com/user-attachments/files/19243114/fc_asm4_lcs.FCStd.zip)

Zolko-123, 2025-03-14:
I can confirm. Quite funny. Doesn't happen with other combinations, so this is a FreeCAD regression

Zolko-123, 2025-05-06:
Any update on this ? Does it still occur ?

Sil68, 2025-05-07:
Tested this with

OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.41573 (Git)
Build date: 2025/05/05 17:45:01
Build type: Release
Branch: main
Hash: 3189030c11010631e726dbe350e9077dc84109bb
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Austria (en_AT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/108/1
Installed mods: 
  * MeshRemodel 1.10.37
  * ThreadProfile 1.96.0
  * Reporting
  * Launcher
  * 3DfindIT 1.2.0 (Disabled)
  * BillOfMaterials 1.0.3
  * woodworking 0.21.2.33771.1.0.1
  * ThreadProfile.backup1741713547.9851232 1.92.0 (Disabled)
  * FreeGrid 2.2.0
  * Defeaturing 1.2.6
  * Gridfinity 0.11.9
  * FEMbyGEN 2.4.0
  * fasteners 0.5.38
  * Lithophane
  * Assembly4.zip
  * lattice2 1.0.0
  * sheetmetal 0.7.22
  * 3D_Printing_Tools
  * FrameForge 0.1.3
  * Assembly4 0.50.18

and all seems to be fine. Was able to reproducibly select two items without fc crashing