Zolko-123/FreeCAD_Assembly4 #552: Colors of objects Mirrored

leoheck, 2025-03-22:
Hi @Zolko-123 , is it normal for replicated Fasteners to have issues with the appearance?

For instance, I have this screw (Gray, normal standard color) on Top-Left, then I mirrored it to the Right and it turned Black.
Then I mirrored both, down... and one is still black, but the second one is the right color.

[image]

If I change the color of the screw... only the 1st and the last change the color.
All seem to change the color, but this Black is hiding the main color.

[image]

leoheck, 2025-03-22:
I was preparing a sample file to show this, but then the fasteners cannot be placed anymore.

Saving the versions here.

- Freecad 1.1.0dev.40655 (Git) Conda AppImage
- Fasteners 0.5.37
- Assembly4 0.50.19-dev

Zolko-123, 2025-03-22:
I guess you're using freecad version 1.1 ? I have read that they introduced some new Material properties and that they conflict with _App::Link_'s `OverrideMaterial` property. On other words, it's a freecad bug/regression

A mirror is a link with a scale of -1

leoheck, 2025-03-22:
Alright, I have a simple example: [mirror_issue.FCStd.zip](https://github.com/user-attachments/files/19403612/mirror_issue.FCStd.zip)

[image]

For polar a pattern, this does not happen.

[image]

leoheck, 2025-03-22:
Saving the versions I am using:

- FreeCad 1.1.0dev.40655 (Git) Conda AppImage
- Assembly4 0.50.19-dev
- fasteners 0.5.37

leoheck, 2025-03-22:
Ah, it happens to Bodies too, not only Fastners.

[image]

leoheck, 2025-03-22:
The linear pattern does not have the same issue.

[image]

Zolko-123, 2025-03-22:
Does this happen with 0.21 and/or 1.0 ?

leoheck, 2025-03-22:
Ah, I see that there is this Property called `Show element`
It toggles this black artifact.

However, since I have nested mirrored patterns It changes the screw that is going to be black but I cannot disable all at once.
Let me show..

leoheck, 2025-03-22:
When I create the first mirrored pattern, the Property is FALSE, but the screw is Black.

[image]

If I toggle it to TRUE, it works (this seems to be an inverted logic)

[image]

Now, if I mirror this, this is what happens

[image]

Toggling this second mirrored pattern to TRUE, this happens

[image]

leoheck, 2025-03-22:
There is something strange. The nested objects change weirdly. Is this expected? 
Check this video.

https://github.com/user-attachments/assets/ae9cc4e5-3503-4bfa-8302-85bddff6e269

leoheck, 2025-03-22:
> Does this happen with 0.21 and/or 1.0 ?

Sorry, I did not see your messages.
Let me check.

leoheck, 2025-03-22:
Yeah, this happens in the 1.0 and also 0.21, 

See, here is in 0.21.

https://github.com/user-attachments/assets/bafb465b-7c4e-4ece-b5ab-39b4844bc32c

leoheck, 2025-03-22:
By toggling the `ShowElement` property, this is the best I can do.

[image]

leoheck, 2025-03-22:
There is something crazy happening.

Changing the `ShowElement` changes the way links are nested and replicated.
Does this make sense?

Both operations with False

[image]

Topmost (second mirrored) with True

[image]

Innermost (first mirrored) with True

[image]

leoheck, 2025-03-22:
I thought that the problem happened because nesting mirror operations could make some overlap.
However, overlapping 2 circular patterns does not replicate the problem.

[image]

leoheck, 2025-03-22:
Also, noticed that when creating a Mirror, it shows this ` Mirror_Screw_i1: Link broken!` twice.

[image]

leoheck, 2025-03-23:
@Zolko-123 the code related to the mirror operation is quite simple. Do you think this is an issue that I have to report to Freecad?

leoheck, 2025-03-23:
The issue happens in this operation here.

[image]

Maybe the negative sign is doing something.

leoheck, 2025-03-23:
Changing this sign to positive, it does not have the same issue, see.
Of course, it does not produce the expected behavior but the black thing is gone.
Is this `scaler` property something from Asm4 or Freecad?

[image]

leoheck, 2025-03-23:
it looks like the sign is inverting the colors as well

leoheck, 2025-03-23:
It looks like when we do th Scale List with a vector with -1,-1,-1 the the black artifact happens.

[image]

If I set it to use always a ScaleVector 1,1,1 the shit does not happen.

[image]

Now, how do I invert the object without using this scale list?

leoheck, 2025-03-23:
This is what we are doing.

[image]

leoheck, 2025-03-27:
@Zolko-123 do you, by any chance, know how to mirror the part placement without scaling with -1? 
Also, I think the operation is mirroring the Part while it thould mirror the Placement only, unless there is a setting for that. 

[image]

Zolko-123, 2025-03-30:
I can confirm that mirrored objects are rendered all black, and this in 0.21 and 1.0 and 1.1 (this one looks a little better:

[image]

[image]

Changing in the link's _View_ to `Override Material` doesn't change the display

leoheck, 2025-03-30:
Good, I put some time to rethink this operation but I couldn't do something that works without scaling to -1. Everyone I ask about this says this is some sort of hack that messes up with the normals and shouldn't be used.

Zolko-123, 2025-03-30:
I found the solution: enable "Backlight" in the Display preferences:

[image]

and then : tadaaaa ....

[image]

leoheck, 2025-03-30:
nice! can you mirrror this one more time using the perpendicular plane to check if the 3rd and 4th copies are going to be good too?

leoheck, 2025-03-30:
Can we set this settings per object? Or at least check if this is enabled to give a msg to the user to enable ot?

leoheck, 2025-03-30:
This post mentions that enabling the backlight ends up "hiding" problematic objects.
https://forum.freecad.org/viewtopic.php?t=69111&start=10

leoheck, 2025-03-30:
The backlight is already enabled on 1.1

[image]

Zolko-123, 2025-03-31:
yes this works also after a second mirror:

[image]

> Can we set this settings per object? Or at least check if this is enabled to give a msg to the user to enable it?

this is a global setting. Checking for this setting when making a mirror would be a good idea, but I don't know how to do that

leoheck, 2025-03-31:
I unfortunately could not make this work on `1.1-40971`

[image]