Zolko-123/FreeCAD_Assembly4 #555: Cannot replace a link when Linked Object property is not empty

leoheck, 2025-04-10:
When I have a broken link, the `Linked Object` property is filled with something like this:

[image]

The `Insert Link` on Asm4 does not offer me the option to fix (`Replace`) the link. See:

[image]

Then if I clear the `Linked Object` property this way:

[image]

I can now `Replace` the link using Asm4 `Insert Link`

[image]

Can we have a better way to detect if the link is broken?

Zolko-123, 2025-04-22:
currently we check whether the LinkedObject is empty or not:

`if selObj.isDerivedFrom('App::Link') and selObj.LinkedObject is None:`

I don't see a property in _App::Link_ which would directly give whether the link is valid or not

leoheck, 2025-04-24:
I did not find anything that helps yet.

Actually, for some reason, I think FReecad thinks the link is from inside the document, and not from an external file.
And this makes the whole process complicated somehow.

Zolko-123, 2025-04-29:
> When I have a broken link, the `Linked Object` property is filled with something like this:
> ...
> Can we have a better way to detect if the link is broken?

in your picture, are you sure that the _link_ is broken ? It looks to me that it's the _part_ that is broken. I can't reproduce a broken link with rubbish info in the `LinkedObject` property : the link is either empty or valid. 

How did you manage to get into that state ?

leoheck, 2025-04-29:
The link is broken and you can reproduce this way.

Insert a part (Asm4_link) to the Assembly that uses a Body from an external file.

Then close the assembly file, rename or move the linked file. 

Open the assembly file again. The link is broken and the property Linked Object of the linked part is empty now.

Now, suppose the user wants to fix the linked object(s) (without knowing Asm4 Insert_link does that) by Updating the link properties directly. Which I did for many items when reorganizing a complex project with many parts.

This will work at that time. However, next time when the Assembly file is opened again the link will be broken again and but it's content will remain. Or at least it will be stored somewhere in the Asm4 (hidden). Maybe a hidden property or something. Report view will report link broken referring to a wrong location or to the previous existing location.

This is off topic but another problem with Freecad is to clean the Linked Object property when the file doesn't exist anymore. This is horrible, since it won't tell the user which Link was there before. On a big project this makes things harder to make it work again with the right part. When an object doesn't exist it doesn't mean the link reference (object link) should be removed.

Zolko-123, 2025-04-29:
yes you're right ... one solution could be to add manually, at creation, that information. For example, we could use the _Label2_ (= _Description_) to store the linked object's file path:

`link.Label2 = link.LinkedObject.Document.FileName`

[image]

leoheck, 2025-04-30:
I like this idea of having a custom property so much.

Maybe the `LinkedObjectDoc` could be the whole path (I saw something like this in another place in FreeCAD):
```python
LinkedObjectPath = "./folder/folder/file.FCStd#Object_Name"
```

Or it could be 2 objects
```python
LinkedObjectDoc = "./folder/folder/file.FCStd"
LinkedObject = "Object_Name"
```

When a link to an object is inserted, we can store the filepath from where it comes (if it is from an external file).
We should use relative paths to the Assembly.
So we can move the project (folder) and share it, and it will work on other computers out of the box.

Also, the Relinking should work using the Insert Link button, but it should also work when updating this property directly.

To relink parts using the `Insert Link`, we can check if the `Linked Object` is blank, *OR* if the `LinkedObjectDoc + LinkedObject` or `LinkedObjectPath` exists.

leoheck, 2025-04-30:
Interesting thing I've noticed now is that the Tooltip of the `Linked Object` property is the Filename of the Linked Object. It is missing the path. But this info should be somewhere. 

Ah, found it. 
This thing already exists.
It is not a relative path, unfortunately (yet).

```
filepath = obj.getLinkedObject().Document.FileName
```