Zolko-123/FreeCAD_Assembly4 #565: LCS - inertial, between two circles

free777cat, 2025-05-01:
LCS - inertial, between two circles - placement depends on the size of the circles

I think this is a bit strange (but maybe that's how it's intended). Maybe add a simple "centering" or adjust the "inertial" work

When I didn't know the behavior of LCS binding between circles, I once got into a very big mess and couldn't understand where the error came from.

[image]

[image]

[image]

Zolko-123, 2025-05-06:
This looks like something about the general attachment engine, not specifically Asm4.

free777cat, 2025-05-07:
[image]
Good time! When I try to install such a bearing through the holes in the housing and the holes on the part (with slightly different diameters), the LCS "runs away" from the middle of the center distance.

closed 2025-05-07 by free777cat