Zolko-123/FreeCAD_Assembly4 #568: Repository Setup

PhoneDroid, 2025-05-26:
If you wouldn't mind I'd suggest adding at least  
`freecad` & `addon` to the repository's topics.

Maybe set the wiki entry as the repository link.
( `https://wiki.freecad.org/Assembly4_Workbench` )

Also to clean it up more, disable 
the unused sections ( packages )

PhoneDroid, 2025-05-28:
There should be a gear icon that opens a  
settings menu to configure the sidebar.

[image]

leoheck, 2025-06-06:
@PhoneDroid, this repo is a mess, and it kind of sucks trying to work with the author.
We should considering forking this shit and continuing the work there. Nobody likes this author on the KiCad Forum anyway.

PhoneDroid, 2025-06-06:
Didn't expect to hear that from the 2nd largest contributor to this project O.o

Zolko-123, 2025-06-06:
> Nobody likes this author on the KiCad Forum anyway

what KiCad forum ?